boschsensortec/Bosch-BSEC2-Library #1: compiling error of basic example

huggymann, 2021-12-08:
Hello, 

when compiling the basic example, i get an error 

`Documents\Arduino\libraries\Bosch-BSEC2-Library\src/bsec2.h:50:10: fatal error: bme68xLibrary.h: No such file or directory
 #include "bme68xLibrary.h"
          ^~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1
Error compiling for board Arduino MKRZERO.
`

I cannot find this file in either the repository nor on my laptop. Have you might not included it in the package?

kartben, 2021-12-08:
It's a bit buried in this doc (https://github.com/BoschSensortec/Bosch-BSEC2-Library/blob/master/examples/bme68x_demo_sample/Quick_Start_Guide.md), but you also need to install `Bosch-BME68x-Library`.

huggymann, 2021-12-08:
thanks man! i didnt see that. I followed the instruction on the main page. solved in  #2

closed 2022-02-11 by huggymann

IamGopinath911, 2022-02-11:
I followed the same procedure and when I ran the bme68x_demo_sample, I encountered the following error in Arduino IDE

Arduino: 1.8.19 (Windows 10), Board: "Adafruit ESP32 Feather, 80MHz, 921600, None, Default"

C:\Users\Gopinath Pugazhendhi\Documents\Arduino\libraries\Bosch-BSEC2-Library-master\examples\bme68x_demo_sample\utils.cpp:42: warning: "_GNU_SOURCE" redefined

 #define _GNU_SOURCE

 

<command-line>: note: this is the location of the previous definition

C:\Users\Gopinath Pugazhendhi\Documents\Arduino\libraries\Bosch-BSEC2-Library-master\examples\bme68x_demo_sample\utils.cpp: In static member function 'static demoRetCode utils::begin()':

utils.cpp:107:8: error: 'dateTimeCallback' is not a member of 'File' {aka 'FsFile'}

  File::dateTimeCallback(dateTime);

        ^~~~~~~~~~~~~~~~

C:\Users\Gopinath Pugazhendhi\Documents\Arduino\libraries\Bosch-BSEC2-Library-master\examples\bme68x_demo_sample\utils.cpp: In static member function 'static bool utils::getFileWithExtension(String&, const String&)':

utils.cpp:162:6: error: 'SdFat' {aka 'class SdFs'} has no member named 'vwd'

  _sd.vwd()->rewind(); /* resets the position of the directory */

      ^~~

utils.cpp:163:27: error: 'SdFat' {aka 'class SdFs'} has no member named 'vwd'

  while (file.openNext(_sd.vwd(), O_READ))

                           ^~~

exit status 1

'dateTimeCallback' is not a member of 'File' {aka 'FsFile'}

falkenbergo, 2022-03-18:
I have the exact same problem. Anyone who can help i would be so glad :)

kurzdev, 2022-05-22:
@IamGopinath911 @falkenbergo Hey, I hope this is still relevant to you but I've managed to identify the issue with the compilation error Gopinath described. It kind of correlates to this problem [here](https://github.com/greiman/SdFat-beta/issues/78). Thus, the solution is to install version 1.1.4 of SdFat - that fixes the issue and allows to compile :smile: