boschsensortec/Bosch-BSEC2-Library #3: example "multi" for Bsec2

huggymann, 2022-02-11:
Hello, could you please make an example for reading multiple bME688 at a time? I hoped the X8 board would do it, but it does read only 1 sensor after the MUX.

huggymann, 2022-03-09:
Hello any comments here?

huggymann, 2022-03-09:
This example could be based on the X8 board. There is an example for X8 board already, which is only reading 1 selectable sensor from the board. But the intention would be with 8 sensors, to read them all :)

huggymann, 2022-05-10:
any news on the multiple readout example?

BST-Github-Admin, 2023-01-03:
Hi @huggymann , this will be added in the next release, where we will also include multi-instance BSEC support.

Deepak-Mvk, 2023-08-30:
Hi @huggymann the multi instance BSEC demonstration is part of [this release](https://github.com/boschsensortec/Bosch-BSEC2-Library/releases/tag/v1.5.2400).

closed 2023-08-30 by Deepak-Mvk