boschsensortec/Bosch-BSEC2-Library #4: esp32s2 Support?

marcusdarmstrong, 2022-02-24:
Hi, I've started working on an indoor air quality monitor and happened to purchase an esp32s2 microcontroller (https://feathers2.io/) rather than the supported original esp32. Unfortunately it appears that the provided ESP32 binary is incompatible with the newer chip—Is there any chance of adding support for the new target?

santaimpersonator, 2022-07-28:
I doubt they would add support for that module, since it doesn't have the BLE functionality required for the [main example](https://github.com/BoschSensortec/Bosch-BSEC2-Library/tree/master/examples/bme68x_demo_sample) of this library.

jesseklm, 2022-09-06:
Created a pull request for ESP32S2: #9

Blandi88, 2022-11-22:
I’m using the esp32s2-tft and I’m having the same problem.

BST-Github-Admin, 2023-01-03:
Hi. v1.3.2200 supports the ESP32-S2, ESP32-S3 and ESP32-C3 variants. If you have any problems, please reopen this issue and add a comment.

closed 2023-01-03 by BST-Github-Admin