boschsensortec/Bosch-BSEC2-Library #5: setTemperatureOffset does nothing

Gnarflord, 2022-02-28:
I'm in the process of porting my application from BSEC to BSEC2 and noticed that the function `setTemperatureOffset` does not work. Doesn't matter if I set an offset of 0, 5 or 50 degrees, the sensor reports the same `BSEC_OUTPUT_HEAT_COMPENSATED_TEMPERATURE`.

So I started comparing the relevant part of both libraries:

`bsec.h`:
```
[...]
    if (_data.status & BME680_NEW_DATA_MSK)
    {
        if (bme680Settings.process_data & BSEC_PROCESS_TEMPERATURE)
        {
            inputs[nInputs].sensor_id = BSEC_INPUT_TEMPERATURE;
#ifdef BME680_FLOAT_POINT_COMPENSATION
            inputs[nInputs].signal = _data.temperature;
#else
            inputs[nInputs].signal = _data.temperature / 100.0f;
#endif
            inputs[nInputs].time_stamp = currTimeNs;
            nInputs++;

            /* Temperature offset from the real temperature due to external heat sources */
            inputs[nInputs].sensor_id = BSEC_INPUT_HEATSOURCE;
            inputs[nInputs].signal = _tempOffset;
            inputs[nInputs].time_stamp = currTimeNs;
            nInputs++;
        }
        if (bme680Settings.process_data & BSEC_PROCESS_HUMIDITY)
[...]
```
and `bsec2.h`:
```
[...]
    if (BSEC_CHECK_INPUT(bmeConf.process_data, BSEC_INPUT_HEATSOURCE))
    {
        inputs[nInputs].sensor_id = BSEC_INPUT_HEATSOURCE;
        inputs[nInputs].signal = extTempOffset;
        inputs[nInputs].time_stamp = currTimeNs;
        nInputs++;
    }
    if (BSEC_CHECK_INPUT(bmeConf.process_data, BSEC_INPUT_TEMPERATURE))
    {
#ifdef BME68X_USE_FPU
        inputs[nInputs].sensor_id = BSEC_INPUT_TEMPERATURE;
#else
        inputs[nInputs].sensor_id = BSEC_INPUT_TEMPERATURE / 100.0f;
#endif
        inputs[nInputs].signal = data.temperature;
        inputs[nInputs].time_stamp = currTimeNs;
        nInputs++;
    }
    if (BSEC_CHECK_INPUT(bmeConf.process_data, BSEC_INPUT_HUMIDITY))
[...]
```

If I simply swap out the new if-expression for the old one the offset is applied!
```
    /* Checks all the required sensor inputs, required for the BSEC library for the requested outputs */
// Doesn't work:
//   if (BSEC_CHECK_INPUT(bmeConf.process_data, BSEC_INPUT_HEATSOURCE))
// Use old expression instead:
    if (bmeConf.process_data & BSEC_PROCESS_TEMPERATURE)
    {
```

07spider70, 2022-05-13:
I have also same problem. I used your solution, thanks a lot!

BST-Github-Admin, 2023-01-03:
Hi @Gnarflord , thanks for the hint. Please check out the latest updates in v1.3.2200 and comment here if the issue still persists.

Deepak-Mvk, 2023-08-30:
This issue is being closed, please re-open if you need any further clarifications.

closed 2023-08-30 by Deepak-Mvk