boschsensortec/Bosch-BSEC2-Library #7: RISC-V core support? Apple Silicon Support?

AussieSusan, 2022-05-31:
There are new devices such as the ESP320C3 that contain a RISC-V core - are there plans to support RISC-V (and the appropriate GCC Risc-V compiler)?
Along similar lines, any support the the Apple Silicon computers?
Susan

awstanley, 2022-06-06:
+1 for RISC-V support.   With the ESP32-C3 being a low powered device well suited to deploying the BME68x devices (with good deep sleep options) it seems strange not to have it supported.

I know it could be a performance limitation on the ESP32-C3 that has kept it off the list, but RISC-V co-processors are being added to other microcontroller products, so it would be helpful even if it were just partial support to provide calibration and/or read and the network the data to a central sensor data handler.

BST-Github-Admin, 2023-01-03:
Hi @AussieSusan , ESP32-C3 support has been added for v1.3.2200. If it doesn't work as expected. Please create a new ticket and report any findings. Thank you.

closed 2023-01-03 by BST-Github-Admin