boschsensortec/Bosch-BSEC2-Library #8: Support for Cortex-m7 or Teensy 4.1

mgfuller333, 2022-07-19:
Hello,

has anyone gotten this to run on the Teensy 4.1 or Cortex m7 processor?

I am unable to compile onto the teensy 4.1.

santaimpersonator, 2022-07-28:
I doubt they would add support for the Teensy 4.1, since it doesn't have the BLE functionality required for the [main example](https://github.com/BoschSensortec/Bosch-BSEC2-Library/tree/master/examples/bme68x_demo_sample) of this library.

BoschSensortec, 2022-08-02:
@mgfuller333 , which IDE are you using to program the Teensy 4.1? With a small change, it is assumed that the cortex-m4f build could be reusable for the m7.

mgfuller333, 2022-08-02:
Arduino. And what small change could I make?

BoschSensortec, 2022-08-02:
@mgfuller333, I assume you are referring to Teensyduino? It is unsure if Teensyduino's builder supports pre-compiled libraries. Assuming it does, in your BSEC2 library directory, create a new directory under `src` named `cortex-m7`. Copy the files from `src/cortex-m4/fpv4-sp-d16-hard` to `src/cortex-m7`.

mgfuller333, 2022-08-07:
[image]
[image]

Darn, still did not work. I really am looking forward to this. Is it possible that Bosch can be certain that it will not work on a Teensyduino builder?