boschsensortec/Bosch-BSEC2-Library #10: ESP8266 freertos not found

dkeekstra, 2022-10-29:
When I try to build the demo for ESP8266 the following error pops up.

`
 Not used: C:\Users\dk\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266SdFat
                 from C:\Users\dk\Documents\Arduino\libraries\Bosch-BSEC2-Library-1.1.2200\examples\bme68x_demo_sample\bme68x_demo_sample.ino:52:
label_provider.h:47:10: fatal error: freertos/FreeRTOS.h: No such file or directory
   47 | #include <freertos/FreeRTOS.h>
`

I tried it on several instalations and serveral versions of the Bosch libraries but no succes.
I applied the adjustmenst suggested by [huggymann](https://github.com/BoschSensortec/Bosch-BSEC2-Library/issues?q=is%3Apr+is%3Aopen+author%3Ahuggymann) but still no succes does some one get the demo building ?

BST-Github-Admin, 2023-01-04:
Hi @dkeekstra , the demo code is not yet intended to be platform agnostic. At the moment, it is meant to only work with the ESP32.

Deepak-Mvk, 2023-08-30:
This issue is being closed, please re-open if you need any further clarifications.

closed 2023-08-30 by Deepak-Mvk