boschsensortec/Bosch-BSEC2-Library #11: Compilation error: text section exceeds available space in board

dadshade12, 2022-11-25:
Not sure what is happening here but when I run the bme68x_demo_sample on a feather esb32 with all the same library versions listed I am getting this error

[image]

BoschSensortec, 2022-12-05:
Hi @dadshade12, ESP32 variant are you using? We have it tested for the Huzzah32 from Adafruit.

dadshade12, 2022-12-05:
I’m using the Huzzah32 feather board from adafruit, I am also un-able to
un-zip the dev kit software file on my computer before I even do anything
with the board it’s self.

victor-costa-avancee, 2023-03-13:
I am having the same problem with the Huzzah32 from Adafruit
```Sketch uses 1403605 bytes (107%) of program storage space. Maximum is 1310720 bytes.```

victor-costa-avancee, 2023-03-14:
I ended up slimming down the sketch to around 473k by removing the bleController and related code for DEMO_BLE_STREAMING_MODE

Deepak-Mvk, 2023-08-29:
This has been tested on the BME 688 development kit and found to be working fine. 
Kindly ensure that the library versions installed are as mentioned on this page - [Quick_Start_Guide](https://github.com/boschsensortec/Bosch-BSEC2-Library/blob/master/examples/bme68x_demo_sample/Quick_Start_Guide.md)