boschsensortec/Bosch-BSEC2-Library #12: How to use prepared AI model (BSEC config file from AI studio) with this library to get gas classification results?

yankobogdan, 2022-12-21:
HI. I've bought a BME688x8 board for Feather and want to create a PoC to distinguish a few smells.
I can use Mac + ESP32 Feather + BME688x8 board to collect data and train models in the AI studio for each particular smell.
AI studio output, in my case, it's a BSEC config file that is clear to me.

But how should I use this BSEC file on ESP32 using Bosh-BSEC2-Library to provide live sensor data and see the results of the AI model work?

Please correct me if I misunderstood the process or was missing something.
I find BME688 and the tech around it very attractive, and such guidance would help me to move forward with testing.

I know it's possible by watching a couple of Youtube videos demonstrating that technology.

Thanks for any support!

BST-Github-Admin, 2023-01-04:
Hi @yankobogdan , you may replace the configuration file that is loaded to the algorithm. Look at this example [line](https://github.com/boschsensortec/Bosch-BSEC2-Library/blob/master/examples/x8_board_examples/basic_config_state/basic_config_state.ino#L32)

Deepak-Mvk, 2023-08-30:
This issue is being closed, please re-open if you need any further clarifications.