boschsensortec/Bosch-BSEC2-Library #24: DRAM segment data does not fit

VitaliyYasko, 2023-04-06:
bme68x_demo_sample.ino.elf section `.dram0.bss' will not fit in region `dram0_0_seg'
region `dram0_0_seg' overflowed by 9240 bytes

I got this every time I try to compile the standard example for any ESP32 board (ESP32 dev board, Adafruit Feather ESP32, Adafruit Feather ESP32 V2, etc.). Lib version 2.0.6.  Arduino IDE 2.0.4, and 1.8.19.

Full output:
Library BSEC2 Software Library has been declared precompiled:
Using precompiled library in /Users/vitaliy/Documents/Arduino/libraries/BSEC2_Software_Library/src/esp32
/Users/vitaliy/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /private/var/folders/ch/86_06ypx61vbycfklx8fp11c0000gn/T/arduino/sketches/2BE8CB973B317B4922E9E8C4D806A760/bme68x_demo_sample.ino.elf section `.dram0.bss' will not fit in region `dram0_0_seg'
/Users/vitaliy/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: DRAM segment data does not fit.
/Users/vitaliy/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: DRAM segment data does not fit.
/Users/vitaliy/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: region `dram0_0_seg' overflowed by 9240 bytes
collect2: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1

ipqate, 2023-04-12:
Hello,

Same for me. Got the same error...any advice?

Regards

VitaliyYasko, 2023-04-12:
> Hello,
> 
> Same for me. Got the same error...any advice?
> 
> Regards

You can manually remove BLE from example, that is definitely not the best solution, but at least you will be able to compile it.

ipqate, 2023-04-14:
Thank you Vitaliy. It solved the problem. 

I do not understand how can it be that the compiled version provided by the manufacturer worked and the same source code compiled, does not:(

Regards.

cpietsch, 2023-04-16:
Same here...
DRAM segment data does not fit.
region `dram0_0_seg' overflowed by 9232 bytes

cpietsch, 2023-04-16:
OK, I found the solution.
You have to use EXACTLY the versions that are mentioned at different locations.

- Arduino 1.8.19 (not the IDE)
- Esp32 v2.0.3
- ArduinoJson (tested for 6.19.4 by Benoit Blanchon)
- RTClib (tested for 2.0.3 by Adafruit)
- SdFat (tested for 2.1.2 by Bill Greiman)

Best is to delete everything i C:\Users\...\Documents\Arduino\libraries and the ESP32 in C:\Users\...\AppData\Local\Arduino15\packages and start fresh.

This will probably also resolve #11

Deepak-Mvk, 2023-08-31:
This has been tested on the BME 688 development kit and found to be working fine.
Kindly ensure that the library versions installed are as mentioned on this page - [Quick_Start_Guide](https://github.com/boschsensortec/Bosch-BSEC2-Library/blob/master/examples/bme68x_demo_sample/Quick_Start_Guide.md)

Deepak-Mvk, 2023-08-31:
This issue is being closed, please re-open if you need any further clarifications.

closed 2023-08-31 by Deepak-Mvk