boschsensortec/Bosch-BSEC2-Library #31: ESP32-S3 Deepsleep implementation

maxdd, 2023-09-24:
Hello,
I would like to use a BME680 sensor with BSEC2 library in a ESP32-S3 with battery.
Can you provide the constraints on using it? BSEC library has quite a lot in terms of timings.
How does the BSEC2 behave?

Deepak-Mvk, 2023-10-11:
Hi @maxdd, The BME680 sensor is supported by the BSEC2 library for IAQ outputs. Please note that the BSEC library always expects an incrementing timestamp to calculate the outputs, so ensure that the time continues to tick during the deepsleep. Currently there is no warning code in BSEC to report if there is any mismatch in the input time.

maxdd, 2023-10-11:
Hello @Deepak-Mvk,
WIth IAQ outputs do you mean it doesn't not support other features? If so can you refer to a specification mentioning this?
In BSEC the esp32 deep sleep implementation was using

```
RTC_DATA_ATTR uint8_t sensor_state[BSEC_MAX_STATE_BLOB_SIZE] = {0};
RTC_DATA_ATTR int64_t sensor_state_time = 0;
```

Is this the intended way?
Are you planning to update the source file here
https://github.com/boschsensortec/BSEC-Arduino-library/blob/master/examples/esp32DeepSleep/esp32DeepSleep.ino

with BSEC2?

Regards,

TCB13, 2024-07-23:
> Is this the intended way?
> Are you planning to update the source file here
> https://github.com/boschsensortec/BSEC-Arduino-library/blob/master/examples/esp32DeepSleep/esp32DeepSleep.ino

Question: did that code ever work? The library seems to use the `millis()` function and that get's reset every time the device goes into sleeping...

https://community.bosch-sensortec.com/t5/MEMS-sensors-forum/Supporting-BME680-with-BSEC-on-ESP32-using-deep-sleep/m-p/10644/highlight/true#M1950

maxdd, 2024-07-25:
If I remember correctly I had to use a gettimestamp function after waking from sleep and connecting to the internet.
As far as I know you need to have the absolute time for the algo to work

TCB13, 2024-07-25:
Thanks for the ideas.

So at some point you did changed this line here https://github.com/BoschSensortec/BSEC-Arduino-library/blob/master/src/bsec.cpp#L457 didn't you?

Because if you didn't then the library will have the wrong time and will mess up the readings...

maxdd, 2024-07-25:
I believe BSEC was aligned to BSEC2 api call so that code might not be valid anymore.
I wouldn't be surprised if they changed and fixed this internally although i ignore how since as you said millis() is getting reset,
What i remember is that if you have time problem, the sensor never go into the proper state after an hour or so of stabilization

TCB13, 2024-07-25:
> I believe BSEC has been aligned to BSEC2 api call so that code might not be valid anymore.

https://github.com/boschsensortec/Bosch-BSEC2-Library/blob/master/src/bsec2.cpp#L254

The BSEC2 code also calls millis()...

> What i remember is that if you have time problem, the sensor never go into the proper state after an hour or so of stabilization

Definitely but from my understanding those time problems come from millis() being reset.

maxdd, 2024-07-25:
This [change ](https://github.com/boschsensortec/BSEC-Arduino-library/commit/5c5c6d1ea22b9de083b709b68dc27277768d79fa) reports that compensation is not supported

> To overcome this the sleep duration can be added to the millis and passed to bsec from the example code.
> Changes in bsec.c is also needed for this to work.

@kegov is there a chance to see a fix here?

I guess bsec.c needs to be somehow reverted to something similar to

https://github.com/boschsensortec/BSEC-Arduino-library/commit/2dd977fd57167bdfbf7753f2c9fadb31ebb9704f#diff-61d5bff668b4530537b21e88250797087f2bad8ccb4aa261551665b293aafb62