boschsensortec/Bosch-BSEC2-Library #32: BSEC2 with rp2040 RPi Pico in VSC

ElVoigtlaendo, 2023-12-13:
HI,

i include in a VSC Project with Platformio the  API BME68x-Library and the BSEC2.
Right now I'm trying with the basic example.

following Error appears by compiling

Linking .pio\build\pico_Core2\firmware.elf
c:/users/XX/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/12.3.0/../../../../arm-none-eabi/bin/ld.exe: cannot find -lalgobsec: No such file or directory
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\pico_Core2\firmware.elf] Error 1

I try the BSEC Library and it works.

I also try the BSEC and BSEC2 in  Arduino IDE and it works too.

i took also the newest BSEC2 Version in lib dictionary. Same Error Message.

Where is the Problem?

platformio.ini
[env:pico_Core2]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = pico
framework = arduino
board_build.core = earlephilhower
lib_deps = 
	boschsensortec/BME68x Sensor library@^1.1.40407
	boschsensortec/BSEC2 Software Library@^1.3.2200

best regards

neffs, 2023-12-14:
I had the same problem. It should be solved with this Pull Request: https://github.com/boschsensortec/Bosch-BSEC2-Library/pull/22

ElVoigtlaendo, 2023-12-15:
thx, it's working. ;)

neffs, 2023-12-15:
Can you leave the issue open? Bosch  don't seem to care about the PRs...