boschsensortec/Bosch-BSEC2-Library #52: region `dram0_0_seg' overflowed by 10696 bytes

Dr-Gandalf, 2024-12-03:
I am having problems to compile the example [/examples/bme68x_demo_sample](https://github.com/boschsensortec/Bosch-BSEC2-Library/tree/master/examples/bme68x_demo_sample)

I followed the instructions from adafruit to compile for the recomended mcu board (Adafruit HUZZAH32)

the following is the result I am geting using platformio Arduino 

```
Processing adafruit_feather_esp32 (platform: espressif32; board: featheresp32; framework: arduino)
--------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/featheresp32.html
PLATFORM: Espressif 32 (6.9.0) > Adafruit ESP32 Feather
HARDWARE: ESP32 240MHz, 520KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20017.0 (2.0.17) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 39 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ArduinoJson @ 6.19.3
|-- RTClib @ 2.1.4
|-- SdFat @ 2.1.2
|-- bsec2 @ 1.8.2610
|-- BME68x Sensor library @ 1.2.40408
|-- ESP32 BLE Arduino @ 2.0.0
|-- SPI @ 2.0.0
|-- Wire @ 2.0.0
Building in release mode
Compiling .pio\build\adafruit_feather_esp32\src\ble_controller.cpp.o
Compiling .pio\build\adafruit_feather_esp32\src\bme68x_datalogger.cpp.o
Compiling .pio\build\adafruit_feather_esp32\src\bsec_datalogger.cpp.o
Compiling .pio\build\adafruit_feather_esp32\src\label_provider.cpp.o
Compiling .pio\build\adafruit_feather_esp32\src\led_controller.cpp.o
Compiling .pio\build\adafruit_feather_esp32\src\main.cpp.o
Compiling .pio\build\adafruit_feather_esp32\src\sensor_manager.cpp.o
Compiling .pio\build\adafruit_feather_esp32\src\utils.cpp.o
Building .pio\build\adafruit_feather_esp32\bootloader.bin
Generating partitions .pio\build\adafruit_feather_esp32\partitions.bin
src/utils.cpp:42: warning: "_GNU_SOURCE" redefined
 #define _GNU_SOURCE

<command-line>: note: this is the location of the previous definition
Compiling .pio\build\adafruit_feather_esp32\lib40c\Wire\Wire.cpp.o
esptool.py v4.5.1
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Compiling .pio\build\adafruit_feather_esp32\libd0f\SPI\SPI.cpp.o
Compiling .pio\build\adafruit_feather_esp32\libe30\Adafruit BusIO\Adafruit_BusIO_Register.cpp.o
Compiling .pio\build\adafruit_feather_esp32\libe30\Adafruit BusIO\Adafruit_I2CDevice.cpp.o
Compiling .pio\build\adafruit_feather_esp32\libe30\Adafruit BusIO\Adafruit_SPIDevice.cpp.o
Compiling .pio\build\adafruit_feather_esp32\lib1f1\RTClib\RTC_DS1307.cpp.o
Archiving .pio\build\adafruit_feather_esp32\lib40c\libWire.a
Compiling .pio\build\adafruit_feather_esp32\lib1f1\RTClib\RTC_DS3231.cpp.o
Archiving .pio\build\adafruit_feather_esp32\libd0f\libSPI.a
Compiling .pio\build\adafruit_feather_esp32\lib1f1\RTClib\RTC_Micros.cpp.o
Compiling .pio\build\adafruit_feather_esp32\lib1f1\RTClib\RTC_Millis.cpp.o
Compiling .pio\build\adafruit_feather_esp32\lib1f1\RTClib\RTC_PCF8523.cpp.o
Compiling .pio\build\adafruit_feather_esp32\lib1f1\RTClib\RTC_PCF8563.cpp.o
Compiling .pio\build\adafruit_feather_esp32\lib1f1\RTClib\RTClib.cpp.o
Compiling .pio\build\adafruit_feather_esp32\libcd9\SdFat\ExFatLib\ExFatDbg.cpp.o
Compiling .pio\build\adafruit_feather_esp32\libcd9\SdFat\ExFatLib\ExFatFile.cpp.o
Compiling .pio\build\adafruit_feather_esp32\libcd9\SdFat\ExFatLib\ExFatFilePrint.cpp.o
Archiving .pio\build\adafruit_feather_esp32\libe30\libAdafruit BusIO.a
Compiling .pio\build\adafruit_feather_esp32\libcd9\SdFat\ExFatLib\ExFatFileWrite.cpp.o
Compiling .pio\build\adafruit_feather_esp32\libcd9\SdFat\ExFatLib\ExFatFormatter.cpp.o
[... 108 lines cut ...]
Compiling .pio\build\adafruit_feather_esp32\FrameworkArduino\esp32-hal-rgb-led.c.o
Compiling .pio\build\adafruit_feather_esp32\FrameworkArduino\esp32-hal-rmt.c.o
Compiling .pio\build\adafruit_feather_esp32\FrameworkArduino\esp32-hal-sigmadelta.c.o
Compiling .pio\build\adafruit_feather_esp32\FrameworkArduino\esp32-hal-spi.c.o
Compiling .pio\build\adafruit_feather_esp32\FrameworkArduino\esp32-hal-time.c.o
Compiling .pio\build\adafruit_feather_esp32\FrameworkArduino\esp32-hal-timer.c.o
Compiling .pio\build\adafruit_feather_esp32\FrameworkArduino\esp32-hal-tinyusb.c.o
Compiling .pio\build\adafruit_feather_esp32\FrameworkArduino\esp32-hal-touch.c.o
Compiling .pio\build\adafruit_feather_esp32\FrameworkArduino\esp32-hal-uart.c.o
Compiling .pio\build\adafruit_feather_esp32\FrameworkArduino\firmware_msc_fat.c.o
Compiling .pio\build\adafruit_feather_esp32\FrameworkArduino\libb64\cdecode.c.o
Compiling .pio\build\adafruit_feather_esp32\FrameworkArduino\libb64\cencode.c.o
Compiling .pio\build\adafruit_feather_esp32\FrameworkArduino\main.cpp.o
Compiling .pio\build\adafruit_feather_esp32\FrameworkArduino\stdlib_noniso.c.o
Compiling .pio\build\adafruit_feather_esp32\FrameworkArduino\wiring_pulse.c.o
Compiling .pio\build\adafruit_feather_esp32\FrameworkArduino\wiring_shift.c.o
Archiving .pio\build\adafruit_feather_esp32\libFrameworkArduino.a
Linking .pio\build\adafruit_feather_esp32\firmware.elf
c:/users/jorge/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\adafruit_feather_esp32\firmware.elf section `.dram0.bss' will not fit in region `dram0_0_seg'
c:/users/jorge/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: DRAM segment data does not fit.
c:/users/jorge/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: DRAM segment data does not fit.
c:/users/jorge/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: region `dram0_0_seg' overflowed by 10696 bytes
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\adafruit_feather_esp32\firmware.elf] Error 1
```

Dr-Gandalf, 2024-12-03:
The same happened when I used the Arduino IDE!!

jr1221, 2024-12-30:
I am just under using **exactly** the dependency versions listed in the README.  This includes ESP32-Arduino version 2.0.3 based on ESP-IDF 3.3.0.