boschsensortec/Bosch-BSEC2-Library #54: updateSubscription Brings an error

forstner-m, 2025-04-02:
Hello,

I am using BSEC2 1.8.2610 with a BME688 under ESP32. I have now tried these functions, but each one returns an error:
bme688.updateSubscription(sensorList, ARRAY_LEN(sensorList), BSEC_SAMPLE_RATE_CONT);
bme688.updateSubscription(sensorList, ARRAY_LEN(sensorList), BSEC_SAMPLE_RATE_LP);
bme688.updateSubscription(sensorList, ARRAY_LEN(sensorList), BSEC_SAMPLE_RATE_ULP_MEASUREMENT_ON_DEMAND);
bme688.updateSubscription(sensorList, ARRAY_LEN(sensorList), BSEC_SAMPLE_RATE_CONT);

Only this function is working:
bme688.updateSubscription(sensorList, ARRAY_LEN(sensorList), BSEC_SAMPLE_RATE_ULP);

Why can the sensor only be operated in ULP mode? This was not the case with the 1.3.2200 version.

Thanks for the help!

Best regards,
Michael

forstner-m, 2025-07-29:
The 1.10.2610 have the same problem.

PaulaScharf, 2026-04-06:
I also get warning code 14 "BSEC_W_SU_SAMPLERATEMISMATCH" when I try sample rates other than ULP. Seems to me, that a specific config has to be given to the sensor. I found configs in this repo [here](https://github.com/boschsensortec/Bosch-BSEC2-Library/tree/master/src/config). Based on their names they seem to be for 3 second or 300 second intervalls. I would love to use a higher interval tho, like 1Hz.
If anybody could give a basic example, to run the sensor at 1Hz I would be very happy.